AcademySoftwareFoundation/rawtoaces #40: Default value and renaming for "exp-comp"

miaoqi, 2017-05-21:
The default value of 1.0 for exp-comp needs to be changed to a higher value. Also, the name of this parameter needs to be changed to “headroom”. Since a headroom factor of 1.0 is not a normal exposure. It was suggested that a default headroom factor of 8 should be used.

aforsythe, 2017-06-27:
This issue was moved to http://www.acescentral.com/t/default-value-and-renaming-for-exp-comp/1150

agentirons, 2017-06-29:
Noticed this issue after making a comment about it on the --aces-gain pull request - I've found that 4.0 is much closer to the default rendering from various commercial apps and the in-camera jpg rendering, testing on both Nikon and Canon images.

agentirons, 2017-10-18:
Considering what I found over in this [acescentral](http://acescentral.com/t/automatic-exposure-compensation-in-raw-processing/1243/4) thread, I wonder if it would in fact make more sense to leave default --headroom at 1.0, knowing now that Canon and Adobe's raw converters are both applying auto exposure compensation on output.

For archival and VFX purposes I'm inclined to want whatever the raw exposure actually is vs. what I thought it was based on a processed jpg, but it's bound to be a confusing issue either way. This option could definitely use a thorough explanation in the rawtoaces documentation, but I don't feel like I even understand enough about what it's doing to help with that yet.

closed 2018-01-05 by miaoqi