AcademySoftwareFoundation/rawtoaces #48: Exposure/White Balance showing dark and green with default settings

agentirons, 2017-05-26:
I am testing out rawtoaces on OS X 10.11.6 using homebrew and the rawtoaces installation guide. I wanted to use raws from cameras that already had included spectral sensitivity data, so I found a Canon 5D Mark II and a Nikon D7000 shot from the following links:

https://www.dpreview.com/reviews/canoneos5dmarkii/33
https://www.dpreview.com/reviews/nikond7000/21

I'm not sure exactly what to expect, but I'm seeing the following results when comparing in After Effects CC 14.1:
[image]

Top row is the D7000, bottom is the 5D, and left side is rawtoaces .exr output while right side is Adobe Camera Raw's output using the Adobe Standard camera profile.

The .exrs were created using `$ rawtoaces _filename.raw_`, no other settings.

They were loaded into After Effects using the OpenColorIO plugin instructions - 32bpcc project, no working space selected, files interpreted as "Preserve RGB", OCIO effect applied using an aces_1.0.3 configuration in Display mode, with Input set to ACES 2065-1 and Output set to sRGB. The .exrs are seen by AE as having an embedded color profile of "ACES Chromaticities".

OS X Finder displays them as what looks like correct white balance but still underexposed:
[image]

aforsythe, 2017-05-26:
@agentirons We've been seeing the same thing with some cameras.  @miaoqi and @scottdyer are trying to track down the bug.

KevinJW, 2017-05-30:
I think you are seeing the results of at least two things here:

The Adobe standard profile does far more artistic adjustments than the RAW to ACES tool is intended to do, so it is not expected to output images that look the same even under ACES RRT+ODT viewing.

Looking at the Finder view - this I think is more correct, so there must be something odd about the After Effects setup you have - I can take those images and process them using an alternative tool also based on libraw and get similar looking images to the Finder image, after applying the RRT+ODT they look white balanced at least

Then you are left with exposure - which without metering the light is hard to judge - using the Q14 patches you could check roughly how linear the conversion is (they are likely 0.1D apart, so taking the log10 of the image should make them have roughly equally spaced values).

Kevin

agentirons, 2017-05-30:
Finder is just doing a linear->sRGB curve, so I can get the same look in After Effects using the nuke-default OCIO config. So in that regard, I'm confused about how different programs treat exposure - if the camera itself, Adobe, and various raw processing programs all agree on one exposure even with various color differences, but the rawtoaces pipeline gives a 2 stop lower exposure, is someone 'right'? I understand there's a lot of subjectivity to this, and that as @KevinJW pointed out the Adobe standard profile is not expected to look like the ACES RRT+ODT, but I assumed that the information contained in the raw data would still have an objective 'as shot' exposure.

As for the green tint, it's hard for me to tell what's going on there. Resolve 14 Beta does the same thing when dropping the exr into an ACES project. It's what makes me wonder if rawtoaces is correctly setting the ACES primaries for these two cameras, since white balance looks normal up until the point that you tell AE or Resolve to interpret the exr file as AP0.

agentirons, 2017-05-30:
[image]

So for comparison, on the left here is what you get by using
`$ rawtoaces --wb-method 1 --mat-method 1 _Canon5Dm2.cr2_`

rawtoaces gave me this output:
`Warning: Cannot find color profile from the RAW, will use the default camera matrix from libraw

The White Balance Coefficients used are: 2.077512 1.000000 2.599043`

I then dropped the exr into AE, added a +4 Exposure effect, then the OCIO aces_1.0.3 plugin with the ACES2065-1 -> sRGB Display transform. This look is much more what I was expecting from the ACES ODT+RRT, in that it's the same exposure as I would get from the Adobe Camera Raw importer or the in-camera JPEG but with some color rendering differences.

KelSolaar, 2017-05-31:
For reference this is the output I get using my own tools, i.e. quite similar to *rawtoaces* in term of exposure: 

[image]

The overexposure and contrast you see in the ACR image is typical from the default ACR tonecurve that shipped with all the Adobe .dcp camera profiles.

aforsythe, 2017-06-02:
@agentirons I think we squished the white balance bug with #57 
Can you test it out?  I also think @KelSolaar is correct regarding the exposure.  You can use --exp-comp to adjust during the conversion, but I think this stems from different cameras using different amounts of headroom for highlights below sensor saturation.  We're exploring ways to handle this more gracefully.

KelSolaar, 2017-06-02:
@aforsythe : Yeah, the Adobe DNG SDK has a `baseline_exposure` tag in that regard. I think it is 0.4 for example in the case of the Mark II and the above photo.

**Baseline Exposure**
```
Description
Camera models vary in the trade-off they make between highlight headroom and shadow noise. Some leave a significant amount of highlight headroom during a normal exposure. This allows significant negative exposure compensation to be applied during raw conversion, but also means normal exposures will contain more shadow noise. Other models leave less headroom during normal exposures. This allows for less negative exposure compensation, but results in lower shadow noise for normal exposures.
Because of these differences, a raw converter needs to vary the zero point of its exposure compensation control from model to model. BaselineExposure specifies by how much (in EV units) to move the zero point. Positive values result in brighter default results, while negative values result in darker default results.
```

agentirons, 2017-06-02:
Thanks, @aforsythe, I can confirm that #57 indeed fixed the white balance for both cameras, using the aces_1.0.3 OCIO config in After Effects or dropping into an ACEScc Resolve 14 Beta project. By adjusting the exposure manually both stills look like they could have come from the same camera as one would expect, so that is a relief to see.

As for exposure, @KelSolaar's description of Baseline Exposure seems to explain a lot of my confusion. Just playing more in AE and Resolve showed that each camera needed different amounts of exposure pushing to match the middle grey color chips to their respective Adobe Standard versions.

Where I'm still a bit lost is in my conceptual framework of how digital camera exposure is supposed to work (and if you want to close this issue and move this to the ACES forums that's fine). I have an antique Kodak Brownie film camera at home, and when I want to shoot with it I know that I can use my Canon 60D to match f/stop, ISO, and shutter speed in order to correctly judge my film exposure. So my impression was that these three settings represented some objective idea of exposure that could be translated across cameras and formats, and tools like Adobe Camera Raw seemed to confirm that. But it sounds like digital capture is a bit more complicated than that. All I can say is that when shooting stills on set, I would assume that after converting to ACES and applying the RRT+ODT that I would see an image approximating what I saw in-camera and know that there's still lots of playing around I can do with the underlying data. I'm okay with being wrong about that but I've yet to read anything about ACES that would tell me otherwise.

Perhaps the spectral sensitivity .json files should include something akin to Adobe's baseline_exposure but with --exp-comp still available as an override?

KelSolaar, 2017-06-02:
There is no guarantee that two camera vendors will use the same exposure metering calibration unfortunately. Using the same exposure settings between two different cameras will get you in the ballpark but if you are expecting to be exact, unfortunately it is game over. Vendors adopts different calibration constants `K`,  [ISO 2720:1974(]https://en.wikipedia.org/wiki/Light_meter#Calibration_constants) does only define a range, that is respected if vendors decide to do so.

> Perhaps the spectral sensitivity .json files should include something akin to Adobe's baseline_exposure but with --exp-comp still available as an override?

Such an attribute would be related to strict camera profiling, which I think makes it too specific for being a direct base level attribute in the JSON schema. Could be included in the `description` attribute maybe?

agentirons, 2017-06-27:
Closing this issue now that white balance is fixed in latest master branch and --aces-gain flag has been added.

closed 2017-06-27 by agentirons