AcademySoftwareFoundation/rawtoaces #51: Invalid illuminants

andyp6, 2017-05-26:
Entering D55 as shown in the example string in the README produces an error:

`$ rawtoaces --wb-method 0 D55 --mat-method 0 input.raw`

```
Error: Unknown light source  -D55.
Please use "--valid-illum" to see list of available light sources.
```

Running --valid-illum produces the following list

```
3200k
cie15-a
cie15-c
cie15-d50
cie15-d55
cie15-d65
cie15-d75
d40
d45
d50
d60
iso7589
```
Is there a difference between cie15-d50 and d50?

aforsythe, 2017-06-02:
We're doing some major cleanup here.  @miaoqi is working on functions for daylight and planckian sources.  We'll then only have to read in sources from files like fluorescent lamps, HMI, etc.

aforsythe, 2017-06-27:
This issue was moved to http://www.acescentral.com/t/invalid-illuminants/1154

miaoqi, 2018-01-05:
@andyp6 This issue should have been addressed in the current version (0.1.0). Can you close it? Thanks!

andyp6, 2018-01-07:
Sure. Sorry I forgot to close this.

closed 2018-01-07 by andyp6