AcademySoftwareFoundation/rawtoaces #67: Building on Windows.

KelSolaar, 2017-06-05:
Hi,

I'm trying to build on Windows and I'm not exactly sure how to pass the various dependencies includes and libraries directories to *CMake* and it is very likely that the various `find_package` macros will not on our system.

Here is my current log:

```
[localhost] local: cmake -DCMAKE_INSTALL_PREFIX=D:\Documents\Development\ThirdParty\rawtoaces/build\release -G "Visual Studio 15 Win64"
-- The C compiler identification is MSVC 19.10.25019.0
-- The CXX compiler identification is MSVC 19.10.25019.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/BuildTools/VC/Tools/MSVC/14.10.25017/bin/HostX86/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/BuildTools/VC/Tools/MSVC/14.10.25017/bin/HostX86/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/BuildTools/VC/Tools/MSVC/14.10.25017/bin/HostX86/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/BuildTools/VC/Tools/MSVC/14.10.25017/bin/HostX86/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Warning (dev) at CMakeLists.txt:11 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "MSVC" will no longer be dereferenced when the policy
  is set to NEW.  Since the policy is not set the OLD behavior will be used.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- D:/Documents/Development/ThirdParty/rawtoaces/build/rawtoaces/lib/openexr/IlmBase/build
CMake Error at lib/CMakeLists.txt:4 (include_directories):
  include_directories given empty-string as include directory.

-- Configuring incomplete, errors occurred!
See also "D:/Documents/Development/ThirdParty/rawtoaces/build/rawtoaces/CMakeFiles/CMakeOutput.log".

Fatal error: local() encountered an error (return code 1) while executing 'cmake -DCMAKE_INSTALL_PREFIX=D:\Documents\Development\ThirdParty\rawtoaces/build\release -G "Visual Studio 15 Win64"'

Aborting.
```

miaoqi, 2017-06-05:
Let me finish a few more core features now and get back to this. 

I did try to generate VS solution before using CMAKE with a similar settings in the configure file. I had the same problem when it comes to IlmBase. I did get a solution file though. Then I needed to open it to continue downloading and compilation for "IlmBase". 

This part of "CMakeLists.txt" should do something:

`include(ExternalProject)
if ( NOT IlmBase_FOUND )
  set (openexr_EXTERNAL_BUILD "${CMAKE_CURRENT_BINARY_DIR}/lib/openexr")
  set (ilmbase_EXTERNAL_BUILD "${CMAKE_CURRENT_BINARY_DIR}/lib/openexr/IlmBase")
  set (ilmbase_EXTERNAL_MAKE_BUILD "${CMAKE_CURRENT_BINARY_DIR}/lib/openexr/IlmBase/build")

  if (WIN32)
      STRING(REGEX REPLACE "\\\\" "/" openexr_EXTERNAL_BUILD ${openexr_EXTERNAL_BUILD})
  	  STRING(REGEX REPLACE "\\\\" "/" ilmbase_EXTERNAL_BUILD ${ilmbase_EXTERNAL_BUILD})
  	  STRING(REGEX REPLACE "\\\\" "/" ilmbase_EXTERNAL_MAKE_BUILD ${ilmbase_EXTERNAL_MAKE_BUILD})
  	  message (STATUS ${ilmbase_EXTERNAL_MAKE_BUILD})

 	  ExternalProject_Add( project_ilmbase
    	GIT_REPOSITORY https://github.com/openexr/openexr.git
    	GIT_TAG "origin/master"

        SOURCE_DIR ${openexr_EXTERNAL_BUILD}
        CONFIGURE_COMMAND cd ${ilmbase_EXTERNAL_BUILD}
      	CMAKE_ARGS -DBuildShared=OFF -DBuildExamples=OFF -DCMAKE_INSTALL_PREFIX=${GLOBAL_OUTPUT_PATH}/project_ilmbase
      	BUILD_COMMAND cmake ${ilmbase_EXTERNAL_BUILD} -G "Visual Studio 14 2015"
      	# BUILD_COMMAND cmake ${ilmbase_EXTERNAL_BUILD}
   	)
   	set (IlmBase_LIBRARY_DIRS "${CMAKE_CURRENT_BINARY_DIR}/project_ilmbase-prefix/src/project_ilmbase-build/Half")
    set (IlmBase_INCLUDE_DIRS "${ilmbase_EXTERNAL_BUILD}/IlmBase/Half")
  else ()
   	ExternalProject_Add( project_ilmbase
    	GIT_REPOSITORY https://github.com/openexr/openexr.git
    	GIT_TAG "master"

    	SOURCE_DIR "${openexr_EXTERNAL_BUILD}"
    	CONFIGURE_COMMAND mkdir "${ilmbase_EXTERNAL_BUILD}/build" && cd "${ilmbase_EXTERNAL_BUILD}/build"
      	CMAKE_ARGS -DBuildShared=ON -DBuildExamples=OFF -DCMAKE_INSTALL_PREFIX=${GLOBAL_OUTPUT_PATH}/project_ilmbase
      	BUILD_COMMAND cmake "${ilmbase_EXTERNAL_BUILD}" && make -s "${ilmbase_EXTERNAL_BUILD}/build"
    )

  endif ()

  set (IlmBase_FOUND TRUE)
endif()`

miaoqi, 2017-06-05:
Getting "Ceres::Solve" built on Windows will be more complicated. Please see the following from its website on Installation 

"
On Windows, we support building with Visual Studio 2010 or newer. Note that the Windows port is less featureful and less tested than the Linux or Mac OS X versions due to the lack of an officially supported way of building SuiteSparse and CXSparse. There are however a number of unofficial ways of building these libraries. Building on Windows also a bit more involved since there is no automated way to install dependencies.
"

I think we can build an "executable",  static library ".lib" or dynamic library ".dll" for the users on Windows and then just distribute them.

KelSolaar, 2017-06-05:
Yeah Ceres is a royal pain, hence my comment on Slack, oh and this was my facebook status earlier:

> Compiling Ceres-Solver on Windows, I want to cry :|

I managed to build a version without _glog_ and _gflags_, I had some nasty compiler errors happening when including _eigen_ otherwise.

KelSolaar, 2017-06-05:
Ideally and whatever the platform is, I would like to be able to pass arguments to *CMake*, something along those lines:

> -DILMBASE_INCLUDE_DIR_HINTS="pat/to/ilmbase/include"

_Ceres Solver_ has actually a very nice configuration in that regard.

KelSolaar, 2017-06-05:
About Ceres on Windows: https://groups.google.com/forum/#!topic/ceres-solver/cR8SSDPFvAk

miaoqi, 2017-06-05:
Probably need to add a few more "FindPackage.cmake" files to get these variables defined so you can pass stuff to CMake. In fact, as your mentioned, "Cere Solver" already has them.

KelSolaar, 2017-06-05:
Yeah, I managed very difficulty to compile Ceres with glog and gflags support yesterday night, everything needs to be dynamically linked among other things.

aforsythe, 2017-06-27:
This issue was moved to http://www.acescentral.com/t/building-on-windows/1145

chistof, 2018-12-03:
Hey guys, is there a compiled version of rawtoaces for Windows somewhere ? I'm about to start the adventure but since I compile things once every 4 years, if someone has done it and can share it, it would be a blast! Thanks!

jacob-danell, 2019-08-21:
Would love to have a windows-build with this but I got it running successfully using WSL on Windows, so we windows users aren't dead in the water at least!
Next thing will be to build a bat file to simply drag the raw files to, to convert them.

bkamphues, 2020-09-10:
Hey there! Just like the other 2 people who already commented on a status update: is there any news on a Windows build?

EmberLightVFX, 2020-09-10:
> 
> 
> Hey there! Just like the other 2 people who already commented on the status of this issue: is there any news on a Windows build?

You can make a build using WSL: https://github.com/ampas/rawtoaces/issues/116

bkamphues, 2020-09-10:
> 
> 
> > Hey there! Just like the other 2 people who already commented on the status of this issue: is there any news on a Windows build?
> 
> You can make a build using WSL: #116

Yeah I know! I got that working perfectly, but we would like to use this tool on our Windows systems without having to use WSL. WSL is not really usable in terms of pipeline development.

KelSolaar, 2020-09-10:
Another alternative is Docker: https://github.com/ampas/rawtoaces/pull/118

bkamphues, 2020-09-10:
> 
> 
> Another alternative is Docker: #118

Ooh thanks! This is perfect! Hadn't even though of it!