AcademySoftwareFoundation/rawtoaces #69: python binding

miaoqi, 2017-06-05:
@KelSolaar , can you help me look into the possibility of creating Python bindings? I prefer Boost.python so we do not need to have another library to depend on. I can tweak C++ APIs to create reasonable python interfaces.

KelSolaar, 2017-06-05:
Sounds good with me, I'll have to think about it a bit.

KelSolaar, 2017-06-09:
Are you planning to build a doc for the C/C++ API, it would help to read through the internals.

aforsythe, 2017-06-09:
I’ve started to move stuff from the README to the wiki.
Would that be a good place to put it, or should it live in the main repo?

Alex

[Δ]<http://www.oscars.org>

Alex Forsythe
Image Engineering Manager

Academy of Motion Picture Arts and Sciences
Science & Technology Council
1313 Vine Street • Hollywood, CA 90028

323.817.4170 • aforsythe@oscars.org<mailto:aforsythe@oscars.org>

KelSolaar, 2017-06-09:
The Wiki is probably a good place for a lot of things, I was thinking more about the technical documentation about the functions, etc... I'm not sure what is the plan for the tool but as @miaoqi was talking about a Python API, I was assuming that it could be used as an exe but also as a library that you can link to which in that case would require a proper C/C++ API documentation.

aforsythe, 2017-06-27:
This issue was moved to http://www.acescentral.com/t/python-binding/1144

antond-weta, 2025-09-14:
It should be fairly easy to add python bindings to the rewritten APIs in `rawtoaces_core.h` and `rawtoaces_util.h`.

I reckon, we should start with the utilities library, as it will be potentially used more widely than the core library, and will give the project more exposure.

I have a bit of experience with `pybind`, but would also like to explore `nanobind` for this project, as it seems to be easier to write and maintain, and also a bit more favourable these days?

hyi18, 2025-09-24:
Looks like this one is still hanging so I would like to give this a go!