AcademySoftwareFoundation/rawtoaces #79: Assertion Failure with Canon 5DS and Sony a7rm2

agentirons, 2017-06-27:
I was testing the new aces-gain flag against various sample images and found two that gave the following error in OS X 10.11.6 using the latest rawtoaces master:

`Assertion failed: (outputBufferSize < 150e6), function configure, file /Users/jonathon/aces_container/aces_Writer.cpp, line 212.
Abort trap: 6`

The two affected cameras are the Canon 5DS (using --wb-method 0 and --mat-method 1) and the Sony a7R II, using [RAW test files](https://www.dpreview.com/reviews/sony-alpha-7r-ii/8) pulled from DP Review. I tried more than one of each to be sure.

miaoqi, 2017-06-27:
Hi, for large files, please use "-h" flag. We could make the file size threshold larger, but we will use "150e6" now. Thanks!

agentirons, 2017-06-27:
Got it, thanks!

closed 2017-06-27 by agentirons