AcademySoftwareFoundation/rawtoaces #86: ACES Container max buffer size - Canon 5D MKII

scoopxyz, 2017-07-23:
**rawtoaces** fails to process Canon 5D MKII images because they are larger than the buffer limitation within ACES Container. Since the CMake is currently pointing to the @miaoqi fork of the project perhaps this should be modified there and then pointed to as the reference source in the README.md instead of the official repo.

This is already referenced in the ACES Containter repo with Scott's issue [ \#4](https://github.com/ampas/aces_container/issues/4)

Example error:
```
~/aces_container/aces_Writer.cpp:206: err aces_Writer::configure(const MetaWriteClip&): Assertion `outputBufferSize < 100e6' failed.
Aborted (core dumped)
```

scoopxyz, 2017-07-23:
I see this was addressed in issue #79 

Though I don't see `-h` as a command line option in the command line help. Changing the title of the issue.

agentirons, 2017-08-24:
It is currently listed this way in `rawtoaces --help`:
`-h                      Half-size color image (twice as fast as "-q 0")`

Since the outputBufferSize error is currently generated by @miaoqi's aces_container project, would it better to mention using the -h flag for large files in the rawtoaces help entry, or to create a filesize check in rawtoaces so that a warning can be generated before it gets to aces_container?

agentirons, 2017-11-28:
Looks like #96 will fix this issue, as the buffer size has already been addressed in the official aces_container repo with miaoqi/aces_container@8ba0982ccfb82c9cc24eb2587bddfc560007b5c3

miaoqi, 2018-01-12:
@scoopxyz Hi Sean, I believe we have increased the size of output buffer in the latest version of "aces_container". Can you verify it and close this issue? Thanks!