AcademySoftwareFoundation/rawtoaces #87: Better documentation for properly adding ltsrc and specsens to tool

scottdyer, 2017-07-27:
If I have my own light source or spectral sensitivities, how do I "install" them so that rawtoaces picks up on them? Obviously I can add them to the data directories, but what is the logic that rawtoaces uses to find a particular camera specsens, for example?

Let's say I have a camera file from a Sony a7S I. The tool does not have specsens for this, but let's say that I do. If I add that json file into the specsens directory, what fields need to be properly formatted for rawtoaces to match the specsens to the camera file?

The logic rawtoaces uses (and thus the required formatting for it to work) needs to be documented for both ltsrc and specsens.

andyp6, 2017-09-15:
It looks like it's the model tag that gets used. I nulled the other fields to test and still get results with custom specsens data. The catalogue number seems like an irrelevant requirement too.

Snehal322, 2026-10-21:
Hi! I'd love to help document how custom camera data gets matched against file headers. I'm planning to draft a documentation update for this.

Snehal322, 2026-10-27:
Hello! I have reviewed issue #87 and successfully set up my development environment for rawtoaces. I would like to take on this issue, investigate the cause, and submit a pull request with a fix soon. Please let me know if there are any specific guidelines I should follow.

soswow, 2026-10-27:
@Snehal322 not really any guidance from the start. All should be covered in .md files ([CONTRIBUTING.md](https://github.com/AcademySoftwareFoundation/rawtoaces/blob/main/CONTRIBUTING.md), [AI_policy.md](https://github.com/AcademySoftwareFoundation/rawtoaces/blob/main/docs/AI_Policy.md)) . show us what you thinking of in draft PR and we go from there