AcademySoftwareFoundation/rawtoaces #91: JSON Schema Validator always gives an error in the license field

andyp6, 2017-09-15:
I get an error Null/String error for the 'license' field when validating JSON with custom and the provided specsens data. It can be ignored but obviously is not validating.

`Message:    Invalid type. Expected String but got Null.
Schema path:    properties/header/properties/license#/type`

miaoqi, 2017-09-15:
Hi, thanks for the feedback. I went to these two site: 

1. https://jsonlint.com/
2. https://jsonformatter.curiousconcept.com/

It looks our current schema and values are valid.

@KelSolaar Hi Thomas, can you take a look? maybe we should need to change it to ""type": ["string", "null"]"

andyp6, 2017-09-16:
I've opened a pull request for this as a license to use spec sens data doesn't appear to be mandatory and I can't see any potential issues.

KelSolaar, 2017-09-18:
Sorry for missing that one!

I would argue that putting a license is mandatory at least from a technical perspective. The first thing a company does when using public open source software (or data) is usually to check the license in order to know if the software can be used. If a software has no license, it is typically avoided like highly dangerous plague :)

andyp6, 2017-09-19:
I don't disagree :)

I was looking at it from the perspective of current JSON, non of which specifies a license and therefore does not validate using the schema provided in README.md

KelSolaar, 2017-09-19:
I think we talked about it already on Slack. Dragging @aforsythe to the convo as it is important :)