AcademySoftwareFoundation/rawtoaces #101: "rawtoaces" does not support passing of RAW file meta data to the resulting ACES file

miaoqi, 2018-01-13:
"rawtoaces" does not pass RAW file meta data to the resulting ACES. The issue is related to the fact that "aces_container" does not support writing the optional attributes to the header. 

To address this issue,

1). update a function call in "aces_container";
2). map the RAW file meta data into ACES's specification

miaoqi, 2018-01-13:
Find the data structure of returned meta data from the RAW by "libraw":

https://www.libraw.org/docs/API-datastruct-eng.html#libraw_iparams_t

We can map members inside "libraw_iparams_t" to the matching optional attributes in ACES specification.

KevinJW, 2018-01-16:
There have, in the past, been some discussions on adding EXIF data to OpenEXR on the development mailing list. You might want to check there as standardisation would be very useful to get vendors to support reading the data correctly. e.g. for HDRI image merging etc.

Kevin

KelSolaar, 2018-01-16:
Yes! I will also chase with Mathias / ILM folks the status.

laserpanda, 2018-09-15:
Any progress? O.o

antond-weta, 2025-08-29:
Closing this. https://github.com/AcademySoftwareFoundation/rawtoaces/pull/167 switches to using OpenImageIO for reading raw files and writing exr files. All raw file metadata which OIIO understands get written out. If there is some specific metadata you need which is not present - file an issue and we'll get that added.