AcademySoftwareFoundation/rawtoaces #102: Bump version number of JSON to 1.0

aforsythe, 2018-01-13:
After all existing JSON issues are closed the version number of the JSON should be bumped up to 1.0.

KelSolaar, 2018-01-13:
Sounds good!

miaoqi, 2018-01-17:
@KelSolaar @aforsythe  where is the latest specification for JOSN 1.0? also what should we do with the "license" field?

KelSolaar, 2018-01-17:
The spec is for now still in the README file: https://github.com/ampas/rawtoaces#json-schema-for-spectral-datasets

I strongly think this is a required/mandatory field, actually even more than last time we talked about it with people here.

miaoqi, 2018-01-19:
Hi, any thoughts for the "license" field?

Thanks!

KelSolaar, 2018-01-20:
@miaoqi  as I said above:

> I strongly think this is a required/mandatory field, actually even more than last time we talked about it with people here.

miaoqi, 2018-01-20:
@KelSolaar I undersand. Any suggestions on what it should be like?

KelSolaar, 2018-01-20:
Oh well, it is Academy duty to decide what is the appropriate license for its data :)

aforsythe, 2018-01-20:
I’m looking into what license requirements make sense

[Δ]<http://www.oscars.org>

Alex Forsythe
Image Engineering Manager

Academy of Motion Picture Arts and Sciences
Science & Technology Council
1313 Vine Street • Hollywood, CA 90028

323.817.4170 • aforsythe@oscars.org<mailto:aforsythe@oscars.org>

Oh well, it is Academy duty to decide what is the appropriate license for its data :)