AcademySoftwareFoundation/rawtoaces #103: Auto Correction of Hot Pixels

agentirons, 2018-01-15:
I can't seem to find any info on whether dcraw has already implemented this, but it would be great to have an option for interpolating around hot pixels in the raw image data automatically, rather than manually using a dark frame for each camera.

[Ufraw](http://ufraw.sourceforge.net/) seems to have added this as a feature, so that may be a good starting point.

miaoqi, 2018-01-17:
I think "libraw" will fix the hot spots before demonsaicing . You can also post a question on libra's forum.

miaoqi, 2018-03-23:
Hi, have you posted a question on "libraw" forum?

agentirons, 2018-07-17:
Closing my issue as I don't have the time to implement the feature, maybe someone can take it up in the future.

closed 2018-07-17 by agentirons