AcademySoftwareFoundation/rawtoaces #104: Assertion fails, core dumps

maxnbk, 2018-03-07:
rawtoaces: [...]/src/rawtoaces/lib/rta.cpp:814: int rta::Idt::loadIlluminant(const std::vector<std::basic_string<char> >&, std::string): Assertion `paths.size() > 0 && !type.empty()' failed.
Aborted (core dumped)

miaoqi, 2018-03-23:
It is possible that the "illuminant" data path is problematic or set up correctly. 

Did you set up your path to the "data"? 

Thanks!

maxnbk, 2018-04-03:
content of my environment variable AMPAS_ILLUMINANT_PATH contains single directory which contains iso7589_stutung_380_780_5.json , and AMPAS_CAMERA_SENSITIVITIES_PATH points to directory which contains arri, canon, nikon, sony jsons.
If I point it to directory one above, same error.

antond-weta, 2025-08-29:
Closing this as resolved, `RAWTOACES_DATA_PATH` works with all 4 data types.

closed 2025-08-29 by antond-weta