AcademySoftwareFoundation/rawtoaces #105: data path not relocateable

maxnbk, 2018-03-07:
in acesrender.cpp, the illuminant and camera data path defaults to "/usr/local/include/rawtoaces/data/illuminant", but a build which uses a CMAKE_INSTALL_PREFIX to be installed elsewhere will not have data there. The default path for the data files should be parameterized by the passed CMAKE_INSTALL_PREFIX if present.

miaoqi, 2018-03-23:
Hi, 

Which OS did you install it on? 

you can also use the environment variable of "AMPAS_DATA_PATH" to set up the path to the data.

Thanks!

maxnbk, 2018-04-10:
So using AMPAS_DATA_PATH works.
Basically same issue as this issue, because the other two env vars have no effect, but data path works for both purposes.
https://github.com/ampas/rawtoaces/issues/104

IdrisMiles, 2018-12-20:
setting `AMPAS_DATA_PATH` does not work for the `cmf` and `training` data.

These are still hardcoded to:
```c++
#define FILEPATH ("/usr/local/include/rawtoaces/data/")
```

Would it be possible to allow these to use `AMPAS_DATA_PATH` or their own unique env var?

antond-weta, 2025-08-29:
Closing this old issue, I assume it has been resolved. `RAWTOACES_DATA_PATH` (which is a replacement for the old `AMPAS_DATA_PATH`) does work with the `cmf` and `training_data`.

closed 2025-08-29 by antond-weta