AcademySoftwareFoundation/rawtoaces #111: Where is the camera spectral data from?

ilia3101, 2019-08-11:
Where did the [camera spectral data](https://github.com/ampas/rawtoaces/tree/master/data/camera) come from, or how was it created? I am interested in getting such data for more cameras but I have no clue about where to start.

I hope this is not too annoying/irrelevant, I'm just really curious.

KelSolaar, 2019-08-16:
It was acquired by the A.M.P.A.S. by @scottdyer using a monochromator among other devices.

scottdyer, 2019-08-16:
Actually, I only captured the data for the Nikon D700 and the two Sony a7 cameras. I'm not sure about the origin of the other data - it may have been provided by Jack Holm or from other experiments by @aforsythe

aforsythe, 2019-08-16:
The D21 came from ARRI. Almost positive the other cameras came from Jack

KelSolaar, 2019-08-16:
Might be worth indicating in the README file or somewhere! :)

ilia3101, 2020-01-08:
Thank you all for the answers. Sorry for leaving the issue open so long with no response. I also wanted to ask: Is there a chance that data for more cameras will be collected by AMPAS some time?

ilia3101, 2020-01-08:
I originally opened this issue because I was experimenting with "IDT"s that use a 2d lookup table in xy space to correct colours a little more precisely than a matrix can. And spectral data is the best way to generate these for any illuminant. Is this of any interest to rawtoaces? I will publish some code for this soon.

antond-weta, 2025-08-29:
Closing this old issue as resolved, as I doubt we'll get any additional info on the topic.