AcademySoftwareFoundation/rawtoaces #116: Get it running on Windows using WSL

jacob-danell, 2019-08-22:
I got rawtoaces running on windows flawlessly using wsl (windows subsystem for linux). I made two batch files, one to install everything and another to convert with.
Until we can get an official windows version, maybe adding these batch files here would help windows users, at least the batch file that helps you convert the file. Or add a link to information on how to install a WSL distro and thus being able to install rawtoaces on windows.

I installed the Ubuntu distro and followed the exact steps as on the front page here except:

- I also install build-essential as I couldn't build anything without it
- I'm downloading aces_container from github.com/ampas/aces_container.git
- I added the curl to download ceres-solver-1.14.0 from ceres's webpage

Thanks for this great software!

[RawToAces.zip](https://github.com/ampas/rawtoaces/files/3529882/RawToAces.zip)

thompsonalt, 2020-08-05:
This saved me a lot of time! Thanks Jacob.

JoeWise, 2020-10-14:
Thanks for this Jacob! Would this work just as well with WSL 2?

Ramel, 2020-10-26:
need to also install "git" and "curl"

mengtaiqi888, 2021-07-18:
[RawToAces.zip](https://github.com/ampas/rawtoaces/files/6836534/RawToAces.zip)
I compiled a version for Windows.

michelerenzullo, 2021-09-22:
> [RawToAces.zip](https://github.com/ampas/rawtoaces/files/6836534/RawToAces.zip)
> I compiled a version for Windows.

Finally! Thank u so much! Could u write how u did it? Because it requires so many dependencies and your build is very light, amazing job. Do you think is possible to create a static lib of ceres? In order to have a single executable. Thanks