AcademySoftwareFoundation/rawtoaces #122: ACES buffer memory leak

cedricp, 2021-01-11:
Hi,
I'm using your library to convert raw footages, it works great, but my computer started to swap after some frames.
I identified that the method **AcesRender::outputACES** does not free the aces buffer. Adding a _delete[] aces_ after **acesWrite()** solved the issue. I think this library deserve a more robust API, I struggled a long time to include/link against it.

BTW, I'm trying to figure out how exposure values are restored from raw data. I can get values > 1.0 in my generated EXT (exaclty what I want, though). Could you explain ?

Best regards

finnschi, 2021-07-12:
I have the same issue

cedricp, 2021-08-19:
Is this project alive ? I really want to figure out why the default scale (gain) is set to 6.0. What are the math behind it ?