AcademySoftwareFoundation/rawtoaces #123: can't intal everything on mac

easdale, 2021-06-29:
Can anyone help me as I've tried installing all the various packages but only home brew and ceres worked. EVERYTHING else has given me errors 'command not found' when pasting into terminal something like  $ brew install rawtoaces

agentirons, 2021-06-30:
Hi @easdale.

"command not found" is odd, as that would mean macOS doesn't know where to find the `brew` command. Just to double-check, are you pasting `$ brew install rawtoaces` or `brew install rawtoaces` into Terminal? What happens if you type `brew --help` into Terminal?

KelSolaar, 2021-07-04:
Do you have Homebrew installed @easdale? https://brew.sh/

riots100, 2021-09-12:
Look like rawtoaces does not build on OSX Big Sur.  
`Error: rawtoaces has been disabled because it does not build!`

soswow, 2025-04-17:
Take a look at latest instructions. I am on Mac and it worked for me