AcademySoftwareFoundation/rawtoaces #124: Full-working release version for Windows, static libs, no DLLs, runtime MD

michelerenzullo, 2021-09-24:
Hi, I enclose the release version for Windows full working with openmp support and with many fix about: FILEPATH, the missing "/" inside "/cmf..." and inside "/training...", path=".", openDir when NULL, memory-leaks after acesWrite. It has been very complicated to compile all the libs with VisualStudio, static libs, and fix the problems with windows. There is another version [here](https://github.com/ampas/rawtoaces/issues/116#issuecomment-882076757) but it has problems with data folder and has some dlls inside. I will publish many details to build later, for now enjoy this. Maybe I will create a repo.

[rawtoaces_WIN_staticlibs_MD.zip](https://github.com/ampas/rawtoaces/files/7228759/rawtoaces_WIN_staticlibs_MD.zip)
.

ktulhu2000, 2023-10-21:
Can we have newer version? Seems it not supports CR3 canon format (

michelerenzullo, 2023-10-21:
I developed my custom new version of rawtoaces in Web assembly based on last libraw and ces_container. Read the help button to see all the different ACES colour spaces outputs and to save in .EXR files 

https://dustfreesolutions.com/single-threaded/weblibraw.html