AcademySoftwareFoundation/rawtoaces #187: Create a GUI application for rawtoaces

antond-weta, 2025-09-07:
A simple GUI application would be nice to have. Ideally implemented with C++ and Qt (using minimal amount of Qt modules).

Minimal functionality, as I see it:
- a list of input files, which can be populated with a file dialog and/or drag and drop,
- checkboxes/dropdowns to configure each parameter,
- a "submit" button and a progress indicator 

Nice to have:
- thumbnails on the input files,
- a simple image viewer to visually configure a crop box and the white-balance box, if enabled,
- some kind of visual error reporting

Amogh2811, 2025-09-07:
Hi! I'm Amogh, a developer exploring open source. I'd love to contribute to this issue . Could you please guide me if any file is currently unassigned? Thanks!

antond-weta, 2025-09-07:
> Hi! I'm Amogh, a developer exploring open source. I'd love to contribute to this issue . Could you please guide me if any file is currently unassigned? Thanks!

Hi Amogh, somebody has started looking into this, as far as I know. I'll ask them to comment here. Perhaps you could work together?

In case you didn't know, the [ASWF dev day](https://www.aswf.io/dev-days/) is coming on 25th September. If you wanted to contribute to rawtoaces, or any other ASWF project, it would be great to see you there. Just pick an issue to work on, which you can finish in one day. This GUI app would be way too big for that.

mikaelsundell, 2025-09-08:
Hi,
I’m designing the new GUI app right now for a draft presentation. @Amogh2811, what type of work are you interested in doing?

Amogh2811, 2025-09-08:
Hi,
My college has assigned me to contribute to an open-source project as part of my coursework, so I’m looking to get involved in something basic related to HTML, Python, or Java. I’m new to this, but I’m eager to learn and contribute wherever I can!

Amogh2811, 2025-09-14:
yes I would love to contribute

antond-weta, 2025-09-14:
> yes I would love to contribute

Great, let's move this discussion to the python bindings [issue](https://github.com/AcademySoftwareFoundation/rawtoaces/issues/69) then.

herronelou, 2026-01-28:
Is there already a draft of the GUI app somewhere in a fork or branch?

antond-weta, 2026-01-28:
We have done some initial work on python bindings, haven't got a full coverage yet, but that should be enough to perform a conversion from python, instead of running a subprocess.

The bindings have not been officially released yet, will probably go live in v2.1, in about a month.